wonho-maker/Adafruit_SH1106 #2: swap() function not defined -> cannot compile

Bzzz, 2016-06-28:
This library misses the swap function from the main SSD1306 library. The example sketch produces:

/libraries/SH1106/Adafruit_SH1106.cpp: In member function 'virtual void Adafruit_SH1106::drawFastVLine(int16_t, int16_t, int16_t, uint16_t)':
/libraries/SH1106/Adafruit_SH1106.cpp:742:23: error: 'swap' was not declared in this scope
       swap(x, y);

SSD1306 library has such a function defined, SH1106 doesn't, maybe carved out by mistake. I propose adding

#define sh1106_swap(a, b) { int16_t t = a; a = b; b = t; }

to the cpp file and replacing every swap() command with sh1106_swap() to stay in line with the main lib.

AphexIce, 2016-11-25:
I also get this error when compiling to do with swap 

Adafruit_SH1106-master\Adafruit_SH1106.cpp:123:14: error: 'swap' was not declared in this scope swap(x, y);
Adafruit_SH1106-master\Adafruit_SH1106.cpp:673:16: error: 'swap' was not declared in this scope swap(x, y);
Adafruit_SH1106-master\Adafruit_SH1106.cpp:741:16: error: 'swap' was not declared in this scope swap(x, y);

ricxcer, 2017-01-09:
How do I do? Can you send me the changed .cpp file? Thank you!

ricxcer, 2017-01-09:
Now it worked! Thanks!

wonho-maker, 2017-01-09:
Hi
It is long time ;; 
Maybe default swap function deprecate or remove in header lib
So i fix it but i can't real test because i don't have arduino now
 
Thank you for everyting

yo1995, 2017-02-16:
thank you for your modification. really helped!

golec-arkadiusz, 2021-01-04:
Edit file Adafruit_SH1106.cpp   from librarry

add function swap(x,y);

//********************** 
int swap(int x, int y)
{
	int tmp;
	tmp = x;
	x = y;
	y = tmp;
	return x,y;
}
//**********************