wonho-maker/Adafruit_SH1106 #8: ESP8266 Support (maybe)

RanTalbott, 2019-02-02:
I made some changes to the driver to get it working on an ESP8266 (Wemos D1 Mini) with I2C OLEDs.

It looks like it's working correctly with my two 1.3" displays, but I've only tested it with the demo program. My only .96 display has a problem with the display being offset 1 or 2 pixels horizontally, and pixels from the rightmost column(s) wrapping around to the left. This is happening on both an Arduino Uno and the D1 Mini, so I think it's something weird about that particular display, not my code.

Since I'm just getting started with OLEDs, I don't have a lot of displays or applications to test with. If you have an AVR-based app that you'd like to try on an ESP8266, or are otherwise interested in doing a little beta testing, drop me a note, and I'll send you a copy of the modified code. If it survives beta testing, I'll offer the changes as suggested updates to this repository.

Two warnings: I haven't tested on an ESP32 yet (waiting on delivery from Banggood), so the setting of the I2C clock rate might be wrong. And I had to sprinkle some yield() calls in the demo routines because some ran long enough that they tripped the watchdog timer. You may have to do the same if you're doing complex graphics.

prof955, 2019-02-08:
> ESP8266

What did you do to pass delay error?

RanTalbott, 2019-02-09:
I haven't noticed any problem with delay(). I'm using version 2.3.0 of the esp8266 board support package. Is there some problem in an earlier or later version I don't know about?

Mikegyver, 2019-08-21:
> Two warnings: I haven't tested on an ESP32 yet (waiting on delivery from Banggood), so the setting of the I2C clock rate might be wrong. And I had to sprinkle some yield() calls in the demo routines [...]

@RanTalbott unfortunately not working on esp32. i have test it just now & got this error.

Arduino: 1.8.9 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

Build options changed, rebuilding all
C:\Users\Hazim Jauhari\Documents\Arduino\libraries\Adafruit_SH1106-master\Adafruit_SH1106.cpp:29:26: fatal error: avr/pgmspace.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

dsyleixa, 2021-05-12:
I can't make the lib compile to target the esp8266 as it uses AVR-specific code, paths, registers, a.s.o.
Can you please provide an esp8266 (and perhaps even esp32-) compatible lib code?