wonho-maker/Adafruit_SH1106 #11: High dynamic memory usage by SH1106 library

LilJohnParadise, 2019-09-01:
Hi, your lib in my program takes 1000 bytes more of dynamic memory than SSD1306 library.

Your library SH1106: The global variable use 1.577 byte (77%) of dynamic memory, leaving 471 byte free for local variables. The max is 2.048 byte.
Low memory available, stability problems may occur.

SSD1306: The global variable use 563 byte (27%) of dynamic memory, leaving 1485 byte free for local variables.

That's because of you not using PROGMEM anywhere in library.

muhit313, 2021-09-13:
I also notice that the library is not  using PROGMEM anywhere. I think it should be modified.

muhit313, 2021-09-24:
@wonho-maker can you try to solve this issue?
Its an major issue for me.

choudharyanil244, 2022-03-21:
I have solve this issue of huge dynamic memory

theintel, 2023-02-19:
> I have solve this issue of huge dynamic memory

Does your forked library resolve the issue?

Andy2No, 2023-02-20:
@theintel You might want to also look at the **Adafruit_SH110x** library.  I compiled one of the examples for an Arduino Uno, which uses nearly 60% of the available code space, while only using 29% of the dynamic memory:

"Sketch uses 18770 bytes (58%) of program storage space. Maximum is 32256 bytes.
Global variables use 614 bytes (29%) of dynamic memory, leaving 1434 bytes for local variables. Maximum is 2048 bytes."

However, it might make more sense to use an Atmega 2560, or an Atmega 1284P, for the extra memory, for use with a graphical display.

theintel, 2023-02-20:
@Andy2No Thanks for that suggestion.

choudharyanil244, 2023-02-20:
> > I have solve this issue of huge dynamic memory
> 
> Does your forked library resolve the issue?

Yes

theintel, 2023-02-20:
@choudharyanil244 Thanks a lot. I'll give it a go.

choudharyanil244, 2023-02-20:
> @choudharyanil244 Thanks a lot. I'll give it a go. 

Sure try it out and let me know the results.

theintel, 2023-02-21:
@choudharyanil244 I tried it out. I wanted to check the real-time dynamic memory usage using freeMemory() following [this tutorial].(https://learn.adafruit.com/memories-of-an-arduino/measuring-free-memory) 

I included `Serial.println(freeMemory())` in different parts of the [sketch](https://gist.github.com/theintel/daa21a97fc8afda21601dee4d923684e) and this was your library's result:  
[image]

And this was the actual library's:
[image]

Sorry for dragging this old thread. I don’t exactly know if this represents optimization performance of the library. So someone's expertise would be helpful.