wonho-maker/Adafruit_SH1106 #15: STM32 board family support

hamedmalekii, 2022-05-03:
hello i need use sh1106 OLED 1.3 DISPLAY to stm32 in ardunio ide when try upload code to my  board i have this error any one can help me>?

CODE: [SELECT ALL](https://www.stm32duino.com/viewtopic.php?f=10&t=1550&p=9834#)

Arduino: 1.8.16 (Windows 10), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STM32duino bootloader, 72Mhz (Normal), Smallest (default)"

C:\Users\hamed\Documents\Arduino\libraries\Adafruit_SH1106-master\Adafruit_SH1106.cpp:40:24: fatal error: util/delay.h: No such file or directory

 #include <util/delay.h>

exit status 1

Error compiling for board Generic STM32F103C series.

This report would have more information with
"Show verbose output during compilation"

albertomercurio, 2022-07-28:
I have also this problem.

hrprada, 2024-11-22:
Same here. No support for SH1106 (spi in my case) onSTM32F103C8 ?