wonho-maker/Adafruit_SH1106 #19: No such file or directory    31 | #include <util/delay.h>

rutuja0325, 2024-02-28:
No such file or directory
   31 | #include <util/delay.h>
   
   i am having this error again and again

vladikas, 2024-07-25:
The library does not know how to work with ESP. Try using a different brunch. https://github.com/wonho-maker/Adafruit_SH1106/pull/16

rin67630, 2024-09-19:
I have the same issue. #16 tells that is does work on ESP8266, but I still do not know how to retrofit util/delay.h
Would it be possible to provide the file in the library?

vladikas, 2024-09-19:
> I have the same issue. 

You're doing something wrong. Check that in the file Adafruit_SH1106.cpp there were lines:
```
#ifdef __AVR__
    #include <avr/pgmspace.h>
    #ifndef __SAM3X8E__
        #include <util/delay.h>
    #endif
#elif defined(ESP8266) || defined(ESP32)
    #include <pgmspace.h>
#endif
```
I tried to compile an example from this library for the ESP8266 board. Everything worked out.

Andy2No, 2024-09-19:
@rin67630 That sort of error normally indicates that you have a different boards support package than the one the code was written for.  Espressif broke compatibility when they changed from 2.x to 3.x.  If you have 3.x installed in Arduino, try uninstalling it then installing v2.0.17, in the boards manager.  If you note the version, you can always put it back later.

I don't know which version this software was meant to support but it's a fair bet that it's not the one you have installed.

rin67630, 2024-09-20:
>  try uninstalling it then installing v2.0.17
I tried with V2.0.0 and with V2.0.1 
Still cannot compile the exampes for a WemosD1 board.
`d:\Activities\3_Maker\libraries\Adafruit_SH1106-master\Adafruit_SH1106.cpp:29:26: fatal error: avr/pgmspace.h: No such file or directory`

Andy2No, 2024-09-20:
@rin67630 The main differences are between 1.x, 2,x and 3.x - Espressif broke backwards compatibility with each major release of its Arduino boards support package, and that tends to show up as missing header files and other components, when trying to compile software written for a different major version.

Andy2No, 2024-09-20:
@rin67630

It seems util/delay.h is meant for AVRs (like the Uno) and defines different versions of delay, but the code here just uses the standard delay(), as far as I can see.  So, try editing out or commenting out that line, in your copy, and see if that produces any more errors.

You should have a definition of delay() from the Espressif board support package - you might want to put that back to the version where you started.  The problem #include doesn't look like it's needed.

vladikas, 2024-09-20:
I'll write again - use another branch. Apparently you are using the original version, which does not support ESP
https://github.com/pedrosousabarreto/Adafruit_SH1106/tree/master